ipab-slmc/exotica #644: Fix header-only implementation of AbstractDDPSolver and OMPLControlSolver

include4eto, 2019-07-23:
`libraryunload` poco exceptions when having the implementation for a base class in a separate `.cpp` file. `AbstractDDPSolver` and `OMPLControlSolver` suffer from it and therefore need to have the implementation in the header file.

@VladimirIvan The other OMPL solver seems to make this work.

TODO: I don't know how or why OMPL solver doesn't suffer from this.

include4eto, 2019-07-23:
See `template class OMPLSolver<SamplingProblem>;` and use 
`planner_allocator_ = boost::bind(&AllocatePlanner<ompl::geometric::LBTRRT>, _1, _2);`-like syntax.

wxmerkt, 2019-07-23:
I think you need to have the template class fully instantiated. Look at the `AbstractTimeIndexedProblem` for an example.

include4eto, 2019-08-19:
Fixed in https://github.com/ipab-slmc/exotica/pull/652

closed 2019-08-19 by include4eto