ipab-slmc/exotica #648: Erroneous output for transformation get functions

cmower, 2019-08-06:
I noticed the transform frames being outputted by `scene.fk('EffFrame', '').get_frame()` (where `EffFrame` is just a constant frame wrt `lwr_arm_7_link`) seems to be incorrect. See image below.

[image]

I am printing to the terminal (bottom) two transform frames. The top (`EffFrame in world_frame (without inverse)`) is that outputted by `scene.fk('EffFrame', '').get_frame()`. Upon inspection, the rotation part is the inverse of what it should be (`EffFrame in world_frame (with inverse)` shows what the transform should be).

wxmerkt, 2019-08-06:
Can you provide a minimally working example?

cmower, 2019-08-08:
Do you mean a potential fix for the problem or an example, in code, of the issue? In the above example, to generate the top transform I (essentially) use 
```python
f1 = scene.fk('EffFrame', '').get_frame()
print f1
```
and in the bottom I use 
```python
f2 = scene.fk('EffFrame', '').get_frame()
f2[:3,:3] = f2[:3,:3].transpose()
print f2
```
The `f1` frame, as printed in the picture above, is the wrong transform frame.

cmower, 2019-08-08:
I looked into this a little deeper and I am certain it is a bug and can explain why. 

Eigen maps the data from the `data` array as such
```c++
int array[9];
for(int i = 0; i < 9; ++i) array[i] = i;
cout << Map<Matrix3i>(array) << endl;
```
and the output is 
```
 0  3  6
 1  4  7
 2  5  8
```
see documentation [here](https://eigen.tuxfamily.org/dox/classEigen_1_1Map.html).

Exotica extracts the rotation matrix from the KDL frame directly from the frames `data` attribute, see [here](https://github.com/ipab-slmc/exotica/blob/cad9fbad9906cefed16cf0d16fbe05dc9c163a23/exotica_core/src/tools/conversions.cpp#L96), and passes it to the Eigen `Map`.

However, KDL stores the rotation matrix as follows (elements are the corresponding indices for illustration).
```
data = {0, 1, 2, 3, 4, 5, 6, 7, 8}

    [ 0 1 2 ]
M = [ 3 4 5 ]
    [ 6 7 8 ]
```
See source [here](http://docs.ros.org/indigo/api/orocos_kdl/html/frames_8hpp_source.html).

wxmerkt, 2019-08-08:
Yeah, another way of doing this is to use `eigen_conversions`, cf. e.g.

https://github.com/ros/geometry/blob/f91b5efc160c40b6d45081fa4a5afc6b829e2059/eigen_conversions/src/eigen_kdl.cpp#L52-L67

cmower, 2019-08-11:
Fixed in #650.

closed 2019-08-11 by cmower