ipab-slmc/exotica #654: DDP, iLQG solvers return wrong-sized solution

wxmerkt, 2019-08-16:
If the problem horizon is `T`, then:

- `X` is of size `T`
- `U` is of size `T-1`
- The `solution` in `Solve` in the solvers is **resized** to `T-1` without initialising or reinitialising to 0. At the end of solve, only the T-1th rows are getting set from the best U. That returns an additional row with a wrongly initialised memory.

Two possible solutions:
a) The returned solution should be of size `T-1`
b) The final command should be a hold/inverse dynamics solution or zeros. 

I tend to (a).

include4eto, 2019-08-16:
So the issue is this line here: https://github.com/ipab-slmc/exotica/blob/3462afd9928c02989754de09229f7ea6daebea23/exotations/solvers/exotica_ddp_solver/src/abstract_ddp_solver.cpp#L61

The solution is resized to `T`, not to `T - 1`, which is the issue (in the first paragraph you said `T-1`).

include4eto, 2019-08-19:
https://github.com/ipab-slmc/exotica/pull/656

PR #656 (2019-09-05): Various Fixes/Issues
- https://github.com/ipab-slmc/exotica/issues/654
- https://github.com/ipab-slmc/exotica/issues/653
- https://github.com/ipab-slmc/exotica/issues/632
- https://github.com/ipab-slmc/exotica/issues/613

Fixes #654, #653, #632, #613

=> Not yet ready for merging, just preview
closed 2019-09-05 by wxmerkt