ipab-slmc/exotica #662: Redundant visuals in KinematicElement

christian-rauch, 2019-09-04:
The `KinematicElement` stores visuals in two ways.

It (1) stores a path to the resource and a shape pointer directly as members:
https://github.com/ipab-slmc/exotica/blob/9e1da605f2b191a17410b2f19be19b5e256c7abc/exotica_core/include/exotica_core/kinematic_element.h#L124-L125
but is (2) also stores an array of `VisualElement`:
https://github.com/ipab-slmc/exotica/blob/9e1da605f2b191a17410b2f19be19b5e256c7abc/exotica_core/include/exotica_core/kinematic_element.h#L129

This appears to be redundant and confusing because it is not documented, if they are used in different ways. Can the members (`shape` and `shape_resource_path`) be removed?

In my scenario, the members are empty and the `VisualElement` actually contains the mesh reference.

wxmerkt, 2019-09-05:
These are from different times and serve different purpose:

- The `shape` is the traditionally loaded collision shape, which gets used in the `CollisionScene`
- The `shape_resource_path` is used when a mesh is loaded via `AddEnvironmentElement` and needs to be used published on the `CollisionShapes` topic because it is not part of the robot kinematics
- The `VisualElement` is for the new visualiser, i.e., MeshCat

I agree documentation and some clean-up once the new visualisers are ready is in order.

christian-rauch, 2019-09-05:
I see that `shape_resource_path` is used for the `visualization_msgs/Marker`. Is the `shape` then only set, when a collision scene if created?

wxmerkt, 2019-09-05:
From what I recall, yes