ipab-slmc/exotica #673: errors triggered in third party libraries

christian-rauch, 2019-10-17:
The recently added error (formerly warning) categories `register` and `ignored-attributes` trigger errors in pybind11:
```
In file included from /usr/include/python2.7/Python.h:88:0,
                 from /opt/ros/melodic/include/pybind11_catkin/pybind11/detail/common.h:112,
                 from /opt/ros/melodic/include/pybind11_catkin/pybind11/pytypes.h:12,
                 from /opt/ros/melodic/include/pybind11_catkin/pybind11/cast.h:13,
                 from /opt/ros/melodic/include/pybind11_catkin/pybind11/attr.h:13,
                 from /opt/ros/melodic/include/pybind11_catkin/pybind11/pybind11.h:43,
                 from [some file that includes pybind11/pybind11.h]:1:
/usr/include/python2.7/unicodeobject.h:534:24: error: ISO C++1z does not allow 'register' storage class specifier [-Werror=register]
     register PyObject *obj,     /* Object */
                        ^~~
/usr/include/python2.7/unicodeobject.h:553:24: error: ISO C++1z does not allow 'register' storage class specifier [-Werror=register]
     register PyObject *obj      /* Object */
                        ^~~
/usr/include/python2.7/unicodeobject.h:575:29: error: ISO C++1z does not allow 'register' storage class specifier [-Werror=register]
     register const wchar_t *w,  /* wchar_t buffer */
                             ^
/usr/include/python2.7/unicodeobject.h:593:23: error: ISO C++1z does not allow 'register' storage class specifier [-Werror=register]
     register wchar_t *w,        /* wchar_t buffer */
```

and Eigen:
```
In file included from /usr/include/eigen3/unsupported/Eigen/CXX11/Tensor:105:0,
                 from [exotica_ws]/src/exotica/exotica_core/include/exotica_core/tools/conversions.h:43,
                 from [exotica_ws]/src/exotica/exotica_core/include/exotica_core/tools.h:37,
                 from [exotica_ws]/src/exotica/exotica_core/include/exotica_core/object.h:33,
                 from [exotica_ws]/src/exotica/exotica_core/include/exotica_core/factory.h:37,
                 from [exotica_ws]/src/exotica/exotica_core/include/exotica_core/task_map.h:39,
                 from [...]:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h: In instantiation of 'struct Eigen::TensorEvaluator<const Eigen::TensorMap<Eigen::Tensor<float, 3, 1> >, Eigen::DefaultDevice>':
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorAssign.h:96:70:   required from 'struct Eigen::TensorEvaluator<const Eigen::TensorAssignOp<Eigen::Tensor<float, 3, 1>, const Eigen::TensorMap<Eigen::Tensor<float, 3, 1> > >, Eigen::DefaultDevice>'
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/Tensor.h:397:14:   required from 'Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>::Tensor(const Eigen::TensorBase<OtherDerived, 1>&) [with OtherDerived = Eigen::TensorMap<Eigen::Tensor<float, 3, 1> >; Scalar_ = float; int NumIndices_ = 3; int Options_ = 1; IndexType_ = long int]'
/usr/include/c++/7/bits/unique_ptr.h:825:30:   required from 'typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = const Eigen::Tensor<float, 3, 1>; _Args = {Eigen::TensorMap<Eigen::Tensor<float, 3, 1, long int>, 0, Eigen::MakePointer>}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<const Eigen::Tensor<float, 3, 1> >]'
[source file using Eigen::Tensor]:   required from here
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h:162:71: error: ignoring attributes on template argument 'Eigen::PacketType<float, Eigen::DefaultDevice>::type {aka __vector(4) float}' [-Werror=ignored-attributes]
     PacketAccess = (internal::unpacket_traits<PacketReturnType>::size > 1),
```

I remember that I got the pybind11 warnings before.
Should we really turn those warnings into errors? I would wait until they are fixed upstream before we activate them in EXOTica.

wxmerkt, 2019-10-17:
I am compiling from source on 18.04 and made sure all dependent package errors are fixed. Are you on the latest master? Do you use gcc or clang and on what operating system? [Eigen Tensor issues are ignored for gcc for sure]

christian-rauch, 2019-10-17:
Yes, I am on master and use gcc. I basically did a git pull and my own code stopped compiling.
The warnings/errors occurred from my own code, e.g. from pybind11 and Eigen headers that I include from my code. I am not sure why they are not triggered in any of the exotica packages. I saw that some warnings/errors are suppressed in the private scope of core packages.

At the moment, I am manually suppressing these errors in my code but it would be useful if the warnings would be suppressed in a global scope.

wxmerkt, 2019-10-17:
Since we had bugs in dependent projects caused by not addressing some warnings, I added the `-Werror` flag to the `exotica.cmake` export, i.e., it will be active on all dependent projects. We can either decide to revert that - or selectively deactivate the warnings when we know an included header violates those errors (e.g., see the pinocchio dynamics solver)

christian-rauch, 2019-10-17:
IMHO configurations of warnings should be kept private to the project unless it is required for compiling/linking against the project. Dependant projects should set their own warnings/errors categories. The same goes for any other compiler flags. Only those that are required to build and link against a library should be exposed.

At the moment, I am fine with deactivating those in my own projects. But it might become annoying in the long run for other projects that use EXOTica. Alternatively, I vote for adding `-Wno-register -Wno-ignored-attributes` to the global `add_compile_options` until the issues are solved upstream.

PR #674 (2019-10-23): [exotica_core] Do not force -Werror for dependent projects
To resolve #673. We should still introduce some of the warnings for dependent projects as they are often correlated to bugs. They can be hidden with pragmas if unfixable.
commit e19a3f57 (2019-10-23): [exotica_core] Do not force -Werror for dependent projects
To resolve #673. We should still introduce some of the warnings for dependent projects as they are often correlated to bugs. They can be hidden with pragmas if unfixable.
closed 2019-10-23 by wxmerkt