ipab-slmc/exotica #708: roslaunch exotica_examples cpp_ik_minimal.launch does not work

mustang66ytz, 2020-04-23:
I am new to exotica and just installed it from source on my Ubuntu 18.04 machine running ROS melodic. There is no error when building from source. However whenever I run any example code, I encounter the same question: segmentation fault.

For example, if I run roslaunch exotica_examples cpp_ik_minimal.launch, there is no error with robot spawning or visualization, the segmentation fault happened within the exotica/exotica_examples/src/xml.cpp. I cannot figure out why, any suggestions will be very helpful!
[image]

wxmerkt, 2020-04-23:
Hi @mustang66ytz,
Thank you very much for your interest in Exotica. I am happy to look into what's going on.

Can you please confirm the commit version you are using and whether you built your workspace in `Debug` or `Release` mode?

You can also add the flag `debug:=true` to a launch file and it opens a separate window with a `gdb` debug session. A stacktrace of the failure could help a lot in debugging this error. Do any of the Python examples work or do all of them fail?

Best wishes,
Wolfgang

mustang66ytz, 2020-04-23:
I am using the master branch, commit 3c40410c2f0ea1c3fa1ae40b5384f3fe39c0e0ee

I built my workspace by executing "catkin build -s", so I guess I did not specify explicitly the build mode?

During debugging, I found out the exception occurs at line 44 
`XMLLoader::Load(file_name, solver, problem);` Where the "file_name" is empty.

So I check the line 42:
` Server::GetParam("ConfigurationFile", file_name);`
where the file_name was assigned by the value of the ROS param `ConfigurationFile`, I can see from my terminal that this ROS param exists and the value is correct, but this `GetParam` did not change the `file_name` variable from empty to non-empty.

Could I know how should I fix this error or hoe should I debug?
The python examples also failed with same error.

Thanks for your suggestions!

wxmerkt, 2020-04-23:
Hi @mustang66ytz,
Thanks for adding the details. It is hard to debug a failure there - the easiest would be if you could post the backtrace of gdb when it fails. If file_name is empty, it could be that it failed to resolve. Are there any local changes to the XML or any other files? Is the workspace sourced correctly?

Cheers

mustang66ytz, 2020-04-23:
Hi @wxmerkt,
There is no change to the local file. Actually I was using the qtcreator to debug the program before. When I run the program with the `deubg` arg set to true, here is the output from my terminal. (I have rebuilt the workspace with debug mode and sourced the bash file).
Thanks for any advice!

```
process[master]: started with pid [3299]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 20a393c6-857a-11ea-af5e-a86daa6c81c1
process[rosout-1]: started with pid [3312]
started core service [/rosout]
RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

xterm -e gdb --args /home/ericyang/exotic_test/devel/lib/exotica_examples/example_cpp_init_xml __name:=example_cpp_init_xml_node __log:=/home/ericyang/.ros/log/20a393c6-857a-11ea-af5e-a86daa6c81c1/example_cpp_init_xml_node-2.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file
[example_cpp_init_xml_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit

```

wxmerkt, 2020-04-23:
Hi @mustang66ytz,
I am trying to reproduce this and cannot. The launch command `roslaunch exotica_examples cpp_ik_minimal.launch` works well. Is it possible that you have multiple copies of Exotica, perhaps also from binary? 

The error from above says that the ConfigurationFile parameter was not given. Which would be different from the initial error (segmentation fault) - in the launch file that should not happen.

Does this command work:
```bash
rosrun exotica_examples example_cpp_init_xml _ConfigurationFile:=$(rospack find exotica_examples)/resources/configs/example_ik.xml
```

Is the output of `rospack find exotica_examples` correct?

Do any of the Python examples work? E.g. python_ik?

wxmerkt, 2020-04-24:
HI @mustang66ytz, does the issue still occur?

wxmerkt, 2020-04-24:
Please reopen if the issue persists.

closed 2020-04-24 by wxmerkt

mustang66ytz, 2020-04-25:
Hi @wxmerkt ,

Thanks for your help! I tried another computer, and it works! Currently I still cannot understand why. I will try to figure it out myself.

Best regards

wxmerkt, 2020-04-25:
Happy to hear it works. If there is anything else I can help you with, please let me know.

mustang66ytz, 2020-04-25:
Thanks very much!
I found the python_ompl_freebase_dubins example very interesting. When I try to launch the example node. I got the following error, where the "/maps" ros topic is not published:

```
[ INFO] [1587809852.423013971]: Loading robot model 'lwr'...
[EXOTica]: No maps were defined!
[EXOTica]: [OMPLDubinsRNStateSpace::SetBounds] Yaw bounds on SE(2) component ignored.
Debug:   Exotica_RRTConnect: Planner range detected to be 8.716616
Info:    Exotica_RRTConnect: Starting planning with 1 states already in datastructure
Info:    Exotica_RRTConnect: Created 5 states (2 start + 3 goal)
Info:    Solution found in 0.004143 seconds
Plotting the solution
```
The solver is able to find solution and plot the graph, but the info to be displayed in rviz is incomplete (no maps or tf). I am new to exotica, not sure where the "maps" are loaded. Could I know what might be the causes for this to happen? Do I need to install or configure anything extra for it to work?

Thanks for your suggestions!
[image]
[image]