ipab-slmc/exotica #710: Obstacle does not display in RViz

jackey984, 2020-05-03:
Thank you for developing this fantastic platform! I should know it earlier and would like to use it in my research project. 

Currently I am playing around the example .launch files. To test OMPL PRM solver, I tried python_prm.launch which worked well. But I want to test without python binding, so modifying the .xml in cpp_ompl.launch:
From: 
https://github.com/ipab-slmc/exotica/blob/80cc22b9717171a18bbf10291e50951532516eb3/exotica_examples/launch/cpp_ompl.launch#L11
To:
```html
<param name="ConfigurationFile" type="string" value="$(find exotica_examples)/resources/configs/example_prm.xml" />
```
In RViz, the robot and path display correctly. However, the box obstacle does not show up. Any comment is appreciated! Thanks!

wxmerkt, 2020-05-03:
Hi @jackey984,
The C++ node advertises the obstacle topic in a different namespace - i.e., if you change the topic of the "SceneObjects" MarkerArray to `/example_cpp_planner_node/CollisionShapes`, you can see the box. 

Do you mind asking me, how did you find out about Exotica?

Any other questions, please let me know.

Best wishes,
Wolfgang

closed 2020-05-03 by wxmerkt

jackey984, 2020-05-03:
Thank you @wxmerkt for your prompt response and fix! Really appreciate!

We are a group of master students working on [TurtleBot3](https://www.robotlab.com/store/robotis-turtlebot-openmanipulator). I am responsible for the motion planning part. We really like your paper "Robust Shared Autonomy for Mobile Manipulation with Continuous Scene Monitoring" and want to try it! We also find out Exotica from that paper. Previously V-REP was used in our team, but lack of flexibility for research purpose. We want to switch to new platform like Exotica in the short future.

wxmerkt, 2020-05-03:
Thank you for the insights and good luck. Please let us know if you have further questions or feedback :-)

PR #711 (2020-05-03): [exotica_examples] Remap C++ visualisation topic to display obstacles in examples
Resolves #710 

cc: @jackey984 This PR resolves the problem you've described by explicitly remapping the topic to the one set in the RViz config.
commit d9dece85 (2020-05-03): [exotica_examples] Remap C++ visualisation topic to display obstacles in examples

Resolves #710