ipab-slmc/exotica #712: [exotica_examples] Python attach example: Goal State is not valid!

the-raspberry-pi-guy, 2020-05-05:
Looking into collisions with EXOTica and launching some of the demos, discovered that the example_attach script has an issue.

Command: 
`roslaunch exotica_examples python_attach.launch`

Error:
```... logging to /home/matt/.ros/log/90fa9f9a-8ece-11ea-9889-448a5ba73575/roslaunch-matt-ubuntu-17502.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://matt-ubuntu:36129/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_description_semantic: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    example_attach_node (exotica_examples/example_attach)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [17512]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 90fa9f9a-8ece-11ea-9889-448a5ba73575
process[rosout-1]: started with pid [17523]
started core service [/rosout]
process[example_attach_node-2]: started with pid [17526]
process[rviz-3]: started with pid [17531]
[ INFO] [1588682844.690521547]: Loading robot model 'lwr'...
[EXOTica]: Warning in Solve: Nominal state regularization is no longer supported - please use a JointPose task-map.
[EXOTica]: Warning in Solve: Nominal state regularization is no longer supported - please use a JointPose task-map.
0
Traceback (most recent call last):
  File "/home/matt/University/Robots/nextage/src/exotica/exotica_examples/scripts/example_attach", line 46, in <module>
    solution.append(ompl.solve().tolist())
RuntimeError: In /home/matt/University/Robots/nextage/src/exotica/exotations/solvers/exotica_ompl_solver/src/ompl_solver.cpp
SetGoalState 122
Object: MySolver: Goal state is not valid!
[example_attach_node-2] process has died [pid 17526, exit code 1, cmd /home/matt/University/Robots/nextage/src/exotica/exotica_examples/scripts/example_attach __name:=example_attach_node __log:=/home/matt/.ros/log/90fa9f9a-8ece-11ea-9889-448a5ba73575/example_attach_node-2.log].
log file: /home/matt/.ros/log/90fa9f9a-8ece-11ea-9889-448a5ba73575/example_attach_node-2*.log
[rviz-3] process has finished cleanly
log file: /home/matt/.ros/log/90fa9f9a-8ece-11ea-9889-448a5ba73575/rviz-3*.log```

wxmerkt, 2020-05-05:
This is related to the recent update in the IK solver. I'll get a fix in for you this afternoon.

the-raspberry-pi-guy, 2020-05-05:
Fixed in wxm-fix-issue-712

closed 2020-05-05 by the-raspberry-pi-guy