ipab-slmc/exotica #722: ParsePath does not work for generated files in a devel workspace

wxmerkt, 2020-08-13:
When a file is generated in a devel workspace, the path resolution is not clear: generated files live in `devel/share/{package_name}`; while rospack will resolve to `src/{package_name}`. Perhaps we need to get all possible paths and check them?

This can happen with URDF files that are generated during build from xacro.

VladimirIvan, 2020-11-05:
The generated URDF files get copied into the SRC where ParsePath should pick it up correctly. Am I missing something here? Or was this a recent change to how xacro functions and it got resolved?

wxmerkt, 2020-11-05:
Not every function call to xacro generates them in `src/` because technically you are not supposed to modify the source directory. The overlay will search both `devel` and `src`. Our ParsePath currently does not do that.