ipab-slmc/exotica #745: Noetic exotica_pinocchio failing to build - cc1plus killed

sloretz, 2021-07-21:
I seem to remember this being a known issue, but I couldn't find where it was talked about. The `exotica_pinocchio` job often fails to build on the Noetic buildfarm. If I remember correctly if the job runs on a machine by itself then it will pass, otherwise it runs out of memory.

https://build.ros.org/job/Nbin_uF64__exotica_pinocchio_dynamics_solver__ubuntu_focal_amd64__binary/82

```
00:10:50.344 c++: fatal error: Killed signal terminated program cc1plus
```

Past work at reducing the memory usage in: #729

wxmerkt, 2021-07-22:
Hi @sloretz,
Indeed I've been seeing these failures quite a lot and was hoping the jobs will eventually pass. There is little more we can do - I split the compilation units down even further (e.g. in 8fb184c0c8226ae5777b4a837ffe909df2237a12) and it didn't help.

One thing that helps - and something I might set as a patch in the next release - is to switch the compiler to clang. I reckon by adding a `<buildtool_depend>clang</buildtool_depend>` and setting it in the `CMakeLists` as a patch that should work - or would there be additional steps required?

sloretz, 2021-07-22:
I didn't find any packages using `clang` to build themselves in Melodic, Noetic, or Rolling, so this package would be the first. As long as the libraries and executables output from clang are ABI compatible with gcc then I would expect it to work with only those two steps.

wxmerkt, 2021-07-23:
I've tested building only one package with clang and the rest with gcc and they are ABI compatible (the Pinocchio package is used as a plugin via pluginlib in the other packages). I've just released a new version to Melodic & Noetic and hope this resolves our memory problems (beyond clang there is nothing further we can do):
- https://github.com/ros/rosdistro/pull/30317
- https://github.com/ros/rosdistro/pull/30318

wxmerkt, 2021-08-23:
Closing as this should be resolved. We still have failures on armhf due to the derivatives of the dynamics in Pinocchio - but we cannot split the compile units any further.

closed 2021-08-23 by wxmerkt