ipab-slmc/exotica #749: pyexotica import error due to missing libmoveit_rdf_loader

cmower, 2022-02-13:
I noticed this error being thrown several times recently. 
```
Traceback (most recent call last):
  File "/home/chris/catkin_ws/devel/lib/ik_ros/ik_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/chris/catkin_ws/src/ik_ros/ik_ros/scripts/ik_node.py", line 9, in <module>
    from ik_ros.single_arm_exotica_effframe import SingleArmExoticaEffFrame
  File "/home/chris/catkin_ws/src/ik_ros/ik_ros/src/ik_ros/single_arm_exotica_effframe.py", line 3, in <module>
    from .exotica import ExoticaBase
  File "/home/chris/catkin_ws/src/ik_ros/ik_ros/src/ik_ros/exotica.py", line 2, in <module>
    import pyexotica as exo
  File "/home/chris/catkin_ws/devel/lib/python3/dist-packages/pyexotica/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: libmoveit_rdf_loader.so.1.1.7: cannot open shared object file: No such file or directory

```

The only fix I've found is to clean and rebuild my catkin workspace from scratch - not ideal!

Any ideas?

wxmerkt, 2022-02-14:
This is likely because you ran `apt-get upgrade` and the new MoveIt version is 1.1.8, but you've linked against 1.1.7. You can check your installed MoveIt version with `dpkg-query -l ros-noetic-moveit-ros-planning`.

cmower, 2022-02-14:
```
$ dpkg-query -l ros-noetic-moveit-ros-planning
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                           Version                      Architecture Description
+++-==============================-============================-============-==========================================
ii  ros-noetic-moveit-ros-planning 1.1.8-1focal.20220201.011651 amd64        Planning components of MoveIt that use ROS
```
Sounds about right, will close for now and reopen if issue persists. Thanks @wxmerkt!

closed 2022-02-14 by cmower