ipab-slmc/exotica #750: Constrain roll and pitch, leave yaw unconstrained

RVSagar, 2022-02-14:
Hi There,

I have a problem where I want to constrain the roll and pitch of an end-effector but leave yaw unconstrained. Is this possible with any of the existing task maps? Perhaps EffOrientation/EffFrame but I was unsure how to set individual Rho values for R, P, Y.  Ultimately I want to use this with a SamplingProblem and plan with RRTConnect on this constraint manifold.

#305 refers to allowing relaxation of individual task map dimensions which got superseded by new task maps, if there is a reference/example to which ones I should use that would be helpful!

Thanks for your help!

wxmerkt, 2022-02-15:
To do this, I would combine `EffPosition` (3d, translation) with an `EffAxisAlignment` (2d, constrains roll and pitch) tasks. However, I don't think this will be successful with a SamplingProblem (which does not work well on constrained manifolds) - and for RRTConnect you'd need a final configuration (e.g. via IK).

I've recently investigated some constrained manifold sampling-based planner within Exotica but the convergence rate is still very slow (~minutes when we'd estimate <1s for optimisation-based). So it may really depend on your task if this is a suitable approach. To make things faster, you can increase the constraint violation tolerance in the sampling problem (Equality tolerance)

RVSagar, 2022-02-15:
@wxmerkt thanks for your reply, much appreciated! I will give that a try.

And thanks for your note on the sampling-based planners with constraints, I figured it would be slower, my experience with other libraries is similar. I essentially want to find a path from Cartesian pose A to B that satisfies the constraints along the path. Would any other problem type from Exotica be good for this (I see some examples that look like trajectory optimization)? Sorry, I am quite new to Exotica but I do like the IK implementation and ease of prototyping :) .

I have two other unrelated questions if you don't mind (I can make new issues if you like):

- Are there any examples of closed-chain planning or IK with Exotica? (e.g., moving an object held by two grippers)
- Is there any notion of 'Cost' for Sampling-Based planners? e.g., planning with RRT Star would require some cost to optimize. I quickly looked through the source but didn't see anything related, only inequality and equality constraints.

Thanks for your help!

wxmerkt, 2022-02-16:
Depending on the robot & the required tasks / environment, trajectory optimisation might be a better approach. I would start with an `UnconstrainedTimeIndexedProblem` in this case with the `AICOSolver` - there are some examples available.

- The humanoid examples perform IK with closed chains (both feet on the ground)
- I have some work for sampling-based planners with custom costs - by default the RRT-Star uses the path length as the cost. This work has not been merged into the public branches yet, but in short - it can be quite slow compared with trajectory optimisation. So it depends on your use case if it's of value or not

RVSagar, 2022-02-16:
Thanks for your answers, I'll take a look at the trajectory optimisation examples and humanoid ones for the closed chain. I'm actually working with a humanoid (PAL REEM-C) and hoping to use Exotica for some wholebody IK tasks. I saw the TALOS example which is useful.

It was mainly curiosity for the RRT Star cost but if I get to needing it I will reach out again.

Thanks for your help, I will close the issue now :)

closed 2022-02-16 by RVSagar